ros-navigation/docs.nav2.org #9: tb3_simulation_launch.py not found error

Marwan99, 2020-04-28:
After following the [getting started](https://navigation.ros.org/getting_started/index.html) installation steps, when running the example step 3 `ros2 launch nav2_bringup tb3_simulation_launch.py` yields an error complaining about not not finding the launch file.

Error message:
`file 'tb3_simulation_launch.py' was not found in the share directory of package 'nav2_bringup' which is at '/opt/ros/dashing/share/nav2_bringup'`

The version of the nav2_bringup package installed form the Ubuntu repository is  0.2.6 and doesn't contain **tb3_simulation_launch.py**, however it contains **nav2_simulation_launch.py** which perform the expected task. So I guess this needs to be changed till the Ubuntu repository package is updated with whats on [github](https://github.com/ros-planning/navigation2/tree/master/nav2_bringup/bringup/launch).

SteveMacenski, 2020-04-28:
`/opt/ros/dashing/share/nav2_bringup` --> dashing. Those instructions are for Eloquent and newer. There's been alot of updates between dashing and eloquent. I highly recommend Eloquent binaries since they're more stable and feature complete until Foxy is released in about a month

Marwan99, 2020-04-28:
Thanks for clarification! I will make the switch to eloquent.

Juts one more question, I had a quick look at the page again I didn't find it mentioning that the example is intended for eloquent, is it worth adding a note about this or did I miss it?  Just to prevent people falling in the same pitfall.

SteveMacenski, 2020-04-28:
I had actually removed any reference to the distribution across the website to intentionally make it easier to grow forward and do less up-keep on the website. I don't think things other than that one have changed appreciably from one distribution to another. 

There's been a conversation about how to do versioning of the documentation (have a different site / version for dashing vs eloquent vs master, read the docs supports) but its unsettled mostly out of lack of interest of the existing developers to go back and retool CI / sphinx

closed 2020-06-10 by SteveMacenski