ros-navigation/docs.nav2.org #21: Updating the parameters in the config guidlines

Marwan99, 2020-06-08:
list of parameters:
- [x] bt_navigator
- [ ] costmaps
- [ ] static_layer plugin
- [ ] inflation_layer plugin
- [ ] obstacle_layer plugin
- [ ] voxel_layer plugin
- [ ] controller_server
- [ ] dwb_local_planner
- [ ] dwb_controller--> publisher
- [ ] oscillation TrajectoryCritic
- [ ] kinematic_parameters
- [ ] xy_theta_iterator
- [ ] 
- [ ]

SteveMacenski, 2020-06-09:
@Marwan99 I'm going to do the lifecycle manager and waypoint follower right now

SteveMacenski, 2020-06-09:
And controller server, because I had a few minutes to kill

Marwan99, 2020-06-09:
@SteveMacenski I will have a go at the bt nodes.

SteveMacenski, 2020-06-09:
My plan today was the planner server and navfn plugin.

SteveMacenski, 2020-06-09:
Also did mapserver

SteveMacenski, 2020-06-09:
Once we're done with all of them, we should remove the tunable parameters page since it will be unnecessary

SteveMacenski, 2020-06-10:
recoveries done... OK done for the day for real now

SteveMacenski, 2020-06-10:
I'll do AMCL tomorrow. You mind taking DWB and the costmaps?

SteveMacenski, 2020-06-10:
Eh, I'll do AMCL right now. Which would you rather do; DWB or costmaps? I'll do the other.

Marwan99, 2020-06-10:
I go with DWB.

SteveMacenski, 2020-06-10:
*phew* I wanted to do costmaps so that's a good fit :laughing:

SteveMacenski, 2020-06-10:
Sweet, we might be done with this by the end of the week. That would be great. I'll post on discourse about it once we're done.

SteveMacenski, 2020-06-10:
@Marwan99 when you're done, remember to delete the existing DWB pages and the "tuning the controller' page

closed 2020-06-12 by SteveMacenski