ros-navigation/docs.nav2.org #39: Manual build instructions missing dependencies

MiguelCompany, 2020-07-09:
When following the instructions [here](https://navigation.ros.org/build_instructions/index.html#manually-build-master) on a fresh install of Ubuntu Focal, I had to install some dependencies that were not mentioned on the instructions.

First step (Build ROS 2 Master) went smoothly.
For the second step (Build Navigation2 Dependencies) I had to add the following dependencies:

```
# install boost
sudo apt install libboost-all-dev
# install gazeebo
curl -sSL http://get.gazebosim.org | sh
```

For the third step (Build Navigation2 Master) I had to add the following dependencies:
```
sudo apt install libgraphicsmagick++1-dev
```

Afterwards, I realized there is a troubleshooting guide, and repeated the process adding the proper call to `rosdep` before doing `colcon build`

Perhaps the best solution would be to add the `rosdep` step to the instructions.

SteveMacenski, 2020-07-09:
Did you use rosdep? Gazebo, Magick, and gazebo are all rosdep entries in the package.xmls (https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml)

MiguelCompany, 2020-07-10:
> Can you submit a PR to add the following before building the dependencies and nav2 workspaces?

Done on #42

PR #42 (2020-07-30): Add rosdep step to manual build instructions
Fix #39
closed 2020-07-30 by SteveMacenski