ros-navigation/docs.nav2.org #147: Planner selector implementation questions

pabloinigoblasco, 2021-03-11:
I show here some questions I have about implementing a convenient PlannerSelector that could be potentially integrated into the navigation stack. 

Goal: the planer selector/goal selector is a "component" that listen to the ROS network (via an action server or a topic). External nodes may ask to switch the planner .

**Note:** This same design principles can be applied to the concept of _GoalSelector_ if the _FollowPath_ action request message is extended with some new _goal_id_ additional field.

In the following code it is shown an example about how the planner selector could be integrated in a navigation-bt.

> <root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <RecoveryNode number_of_retries="6" name="NavigateRecovery">
          <PipelineSequence name="NavigateWithReplanning">
          <PlannerSelector default_controller_id="FollowPath" default_planner_id="GridBased"/> 
             <RateController hz="0.1">
              <ComputePathToPose goal="{goal}" path="{path}"/>
              </RateController>
            <FollowPath path="{path}"/>
        </PipelineSequence>
      <ReactiveFallback name="RecoveryFallback">
        <GoalUpdated/>
      </ReactiveFallback>
    </RecoveryNode>
  </BehaviorTree>
</root>

**Description**
- The PlannerSelector listen to the ROS network to get the "selected planner and controller". 
- The PlannerSelector is "updated" at the beginning of the "navigation cycle" and writes into the blackboard the selected planner and controller (the implementation could use input and output ports but using the blackboard is a more integrated approach). 
- Then the ComputePathToPose and the FollowPath BTNodes reads from the blackboard to fill with that information the action request messages (foor the planner_server and the controller_server)

**Previous experiments**
I tried to implement the concept explained above. The ROS network communication mechanism used was "node parameters". The PlannerSelector had a _PlannerSelector::onParametersUpdated_ callback and checked if the "PlannerSelector.planner_id" and/or "PlannerSelector.controler_id" where updated.

However I got some weird behavior. It looked like many of the parameter updates "were lost". 
This was my interpretation about what happened (Please correct me if you think I am wrong): The lifetime of the PlannerSelector is short. Because of that the _PlannerSelector::onParametersUpdated_ was not called.
(I have doubts about this)

**Workaround**
Since that first approach failed I used a Decorator instead of a regular BTNode.

Already exist a PlannerSelector experiment in the following fork ([PlannerSelector.hpp](https://github.com/pabloinigoblasco/navigation2/blob/feature/smacc2_rolling_stable/nav2_behavior_tree/include/nav2_behavior_tree/plugins/decorator/planner_selector.hpp) and [PlannerSelector.cpp](https://github.com/pabloinigoblasco/navigation2/blob/feature/smacc2_rolling_stable/nav2_behavior_tree/plugins/decorator/planner_selector.cpp))

This solution (fork [https://github.com/pabloinigoblasco/navigation2](https://github.com/pabloinigoblasco/navigation2)): works properly using parameters updates from the terminal like it is shown below: 

> ros2 parameter set /bt_navigator PlannerSelector.controller_id "MyController"

or 

> ros2 parameter set /bt_navigator PlannerSelector.planner_id "MyPlanner"

SteveMacenski, 2021-03-11:
https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/decorator/goal_updater_node.cpp the goal checker is very similar to what we're talking about here (topic in -> change some output port -> input port to another), this was tested and utilized in the follow_path.xml dynamic object following demos @fmrico conducted!

https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/action/compute_path_to_pose_action.cpp#L34 the algorithm Bt nodes use ports already for selecting IDs so using the blackboard variable ports seems like a very simple solution to the issue of conveying information between the BT nodes (that was conveyed from a topic/service via ROS2)

I think we've covered some of the other important topics here on the slack thread - any additional things we should cover as part of this?

pabloinigoblasco, 2021-03-11:
That is the correct reference example. Thanks.
- So I will introduce the following changes to the initial experiment:
    - using i/o ports instead hardcoded-propagation via blackboard
    - replacing the usage of properties with (maybe services or actions instead of topics)

**Additional issue**
- Is a topic subscription a good solution for the planner_selector and the goalchecker_selector?

It may be good to respond some feedback to the requester for example in the case the planner, controller or goalchecker is unknown  or it is not registered.

- Is it convenient conceptually separating planner_selector and controller_selector? Right now planner_selector aggregates the concept of (planner_selector and controller_selector)

- An minor priority question. It is just curiosity. It  looks like this is a design pattern (planner_selector, goalchecker_selector, maybe controller_selector). Shouldn't we use a template base class to let the user implement their own inputs into the nav-tree?
- Possible details of the base class:
    - c++ template parameters: TOutput (ie: std_msgs::msg::String, geometry_msgs::msg::PoseStamped), optional TInput
    - output port name string
    - optional input port name string
    - 
Maybe that is overengineering, but may be interesting to know what you think.
    
Regards.

SteveMacenski, 2021-03-11:
>   Is a topic subscription a good solution for the planner_selector and the goalchecker_selector?

Topic seems reasonable

> Is it convenient conceptually separating planner_selector and controller_selector? Right now planner_selector aggregates the concept of (planner_selector and controller_selector)

I think so, since if we have other BT nodes like coverage planning servers, or route planning servers, that 1 node would really start to blow up. Separating them makes them more easily testable and scale as the project gets more complex.

pabloinigoblasco, 2021-03-15:
I am implementing around the planner_selector, controller_selector and goal_selector.
I am totally miming the goal_updater. That means that we will inherit:
- usage of decorator node
- configurable input topic name

Anything else to say about this?

BTW, I guess it is also a good moment to also check if the naming is good. Now they are being implemented as "PlannerSelector", "ControllerSelector" and "GoalSelector".

I keep this previous plan, but:
Everything okay with that? Maybe PlannerSwitcher? PlannerConfiguration?

SteveMacenski, 2021-03-15:
I'm happy with all of those, "Selector" name, topic names, port names. Yes, I think a `default_planner_id` just as the compute path to pose and follow path have would be good. That way it returns those values when none is set and the `planner_id={selected_planner_id}` still would enable default behavior

pabloinigoblasco, 2021-03-16:
I created an initial PR for the planner_selector before applying PRs for others (controller_selector and goal_checker_selector)

https://github.com/ros-planning/navigation2/pull/2249

SteveMacenski, 2021-03-16:
Closing, see Nav2 ticket / PR set

closed 2021-03-16 by SteveMacenski

Lannister-Xiaolin, 2022-05-31:
Any one can share a xml configure that works for galactic  version? I setted as below, it's not working!!!
```
<root main_tree_to_execute="MainTree">
    <BehaviorTree ID="MainTree">
        <RecoveryNode number_of_retries="6" name="NavigateRecovery">
            <PlannerSelector selected_planner="{selected_planner}" default_planner="GridBased"
                             topic_name="planner_selector">
                <PipelineSequence name="NavigateWithReplanning">
                    <RateController hz="1.0">
                        <RecoveryNode number_of_retries="1" name="ComputePathToPose">
                            <ComputePathToPose goal="{goal}" path="{path}"/>
                            <ReactiveFallback name="ComputePathToPoseRecoveryFallback">
                                <GoalUpdated/>
                                <ClearEntireCostmap name="ClearGlobalCostmap-Context"
                                                    service_name="global_costmap/clear_entirely_global_costmap"/>
                            </ReactiveFallback>
                        </RecoveryNode>
                    </RateController>
                    <RecoveryNode number_of_retries="1" name="FollowPath">
                        <FollowPath path="{path}" controller_id="FollowPath"/>
                        <ReactiveFallback name="FollowPathRecoveryFallback">
                            <GoalUpdated/>
                            <ClearEntireCostmap name="ClearLocalCostmap-Context"
                                                service_name="local_costmap/clear_entirely_local_costmap"/>
                        </ReactiveFallback>
                    </RecoveryNode>
                </PipelineSequence>
            </PlannerSelector>
            <ReactiveFallback name="RecoveryFallback">
                <GoalUpdated/>
                <RoundRobin name="RecoveryActions">
                    <Sequence name="ClearingActions">
                        <ClearEntireCostmap name="ClearLocalCostmap-Subtree"
                                            service_name="local_costmap/clear_entirely_local_costmap"/>
                        <ClearEntireCostmap name="ClearGlobalCostmap-Subtree"
                                            service_name="global_costmap/clear_entirely_global_costmap"/>
                    </Sequence>
                    <Spin spin_dist="1.57"/>
                    <Wait wait_duration="5"/>
                    <BackUp backup_dist="0.15" backup_speed="0.025"/>
                </RoundRobin>
            </ReactiveFallback>
        </RecoveryNode>
    </BehaviorTree>
</root>
```