ros-navigation/docs.nav2.org #172: Tutorial: Navigate through poses

SteveMacenski, 2021-06-03:


FelixZF-Happy, 2026-02-03:
Hi Steve,

I’m a robotics software developer based in Toronto, Canada, with 3+ years of industry experience, and I’ve previously worked with Navigate Through Poses. However, I don’t have access to industrial environment simulations, so I’m wondering whether there are any additional open-source simulation environments that could be quickly deployed. My goal is to use them to build practical tutorials for the community.

I’ve also been following the long-standing issue #5446 (continuing from #4789 and #4593). When I have time in recent 2-3 months, I plan to look into how this feature was completed, and hopefully test it once it is merged from ROS 2 Rolling into the other distributions.

Best regards,
Felix

FelixZF-Happy, 2026-02-03:
Hi Steve,

Yes — I do have access to a physical robot. I have a TurtleBot3 at home (thanksfully I didn't sell it to Italy two years ago), so I can use it for real-robot demonstrations and documentation.

That sounds great. I’ll take it out from the basement and start working on this.

Please let me know if there’s a Slack channel or any preferred place to report progress or share updates.

Best regards,
Felix

FelixZF-Happy, 2026-02-03:
That makes sense — thank you for the clarification, Steve! I agree that reproducibility is important, and using simulation is the right approach for a tutorial that others can easily follow.

I’ll focus on the simulation setup and align it with the TB4-style commercial environment you mentioned.

Best regards,
Felix

SteveMacenski, 2026-02-03:
Great! ❤️ Thanks so much for answering the call of docs! It takes a village!

FelixZF-Happy, 2026-02-23:
Hi Steve,

This is my first time contributing to an open-source repository like this — thank you for the opportunity.

This week I finally had some time to focus on it. The tutorial required quite a bit of reading on my side, including the Behavior Tree documentation and Nav2 docs. I also reviewed some related discussions on Reddit and Stack Overflow to collect common issues other users encountered, meanwhile I learned a few new tools.

I apologize for the mistakes in my two pull requests closed. I’ve learned from them and won’t repeat those issues again. If needed, please feel free to delete them permanently.

**I updated the Navigate Through Poses Behavior Tree documentation as well, since the new BT structure is already reflected on the website but parts of the description seemed slightly outdated.**

I also noticed that the robot_navigator.py changes in the Rolling branch have not yet been synchronized to Kilted or Jazzy. When running the Python script with goThroughPoses function, it fails at the logging step due to this mismatch.

If there are any mistakes or areas that need improvement, please feel free to comment — I’d be happy to continue refining it.

Thank you again for giving me this opportunity.

Best regards,
Felix