ros-navigation/docs.nav2.org #210: First-Time Robot Setup Guide: No map received

chargerKong, 2021-09-18:
I'm reading the tutorial of how to setup the simulated robot.
All things are OK until I run [this](https://navigation.ros.org/setup_guides/sensors/setup_sensors.html#launching-slam-toolbox)
```
ros2 launch slam_toolbox online_async_launch.py
```

The terminal's info
```
[INFO] [launch]: All log files can be found below /home/kong/.ros/log/2021-09-18-17-52-47-576626-r9000x-2021-34874
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [34877]
[async_slam_toolbox_node-1] [INFO] [1631958768.060649413] [slam_toolbox]: Node using stack size 40000000
[async_slam_toolbox_node-1] [INFO] [1631958768.087470152] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-1] [INFO] [1631958768.087805046] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[async_slam_toolbox_node-1] Info: clipped range threshold to be within minimum and maximum range!
```

Here is what the Rviz shows
[image]

If necessary, here is the display.launch.py
```
  1 import launch                                                                                                                                        
  2 from launch.substitutions import Command, LaunchConfiguration
  3 import launch_ros
  4 import os
  5 
  6 def generate_launch_description():
  7     pkg_share = launch_ros.substitutions.FindPackageShare(package='sam_bot_description').find('sam_bot_description')
  8     default_model_path = os.path.join(pkg_share, 'src/description/sam_bot_description.urdf')
  9     default_rviz_config_path = os.path.join(pkg_share, 'rviz/urdf_config.rviz')
 10     world_path = os.path.join(pkg_share, 'world/my_world.sdf')
 11 
 12     robot_state_publisher_node = launch_ros.actions.Node(
 13         package='robot_state_publisher',
 14         executable='robot_state_publisher',
 15         parameters=[{'robot_description': Command(['xacro ', LaunchConfiguration('model')])}]
 16     )
 17     joint_state_publisher_node = launch_ros.actions.Node(
 18         package='joint_state_publisher',
 19         executable='joint_state_publisher',
 20         name='joint_state_publisher',
 21         condition=launch.conditions.UnlessCondition(LaunchConfiguration('gui'))
 22     )
 23     robot_localization_node = launch_ros.actions.Node(
 24        package='robot_localization',
 25        executable='ekf_node',
 26        name='ekf_filter_node',
 27        output='screen',
 28        parameters=[os.path.join(pkg_share, 'config/ekf.yaml'), {'use_sim_time': LaunchConfiguration('use_sim_time')}]
 29     )
 30     rviz_node = launch_ros.actions.Node(
 31         package='rviz2',
 32         executable='rviz2',
 33         name='rviz2',
 34         output='screen',
 35         arguments=['-d', LaunchConfiguration('rvizconfig')],
 36     )
 37
 38     spawn_entity = launch_ros.actions.Node(
 39         package='gazebo_ros',
 40         executable='spawn_entity.py',
 41         arguments=['-entity', 'sam_bot', '-topic', 'robot_description'],
 42         output='screen'
 43     )
 44     
 45     return launch.LaunchDescription([
 46         launch.actions.DeclareLaunchArgument(name='gui', default_value='True',
 47                                             description='Flag to enable joint_state_publisher_gui'),
 48         launch.actions.DeclareLaunchArgument(name='model', default_value=default_model_path,
 49                                             description='Absolute path to robot urdf file'),
 50         launch.actions.DeclareLaunchArgument(name='rvizconfig', default_value=default_rviz_config_path,
 51                                             description='Absolute path to rviz config file'),
 52         launch.actions.DeclareLaunchArgument(name='use_sim_time', default_value='True',
 53                                             description='Flag to enable use_sim_time'),
 54         launch.actions.ExecuteProcess(cmd=['gazebo', '--verbose', '-s', 'libgazebo_ros_factory.so', world_path], output='screen'),
 55         joint_state_publisher_node,
 56         # joint_state_publisher_gui_node,
 57         spawn_entity,
 58         robot_state_publisher_node,
 59         robot_localization_node,
 60         rviz_node
 61     ]) 
```

SteveMacenski, 2021-09-27:
@arubenecia can you look into this?

arubenecia, 2021-10-04:
We also encountered "No map received" problem on the Foxy version of slam-toolbox. We have resolved the issue by updating to Galactic version.

closed 2021-11-29 by SteveMacenski