ros-navigation/docs.nav2.org #244: Add images / gifs to configuration pages for demos

SteveMacenski, 2021-12-01:
- RPP
- Map server
- AMCL
- DWB
- NavFn
- Costmap 2D
- Waypoint follower

NeoKish, 2022-03-01:
@SteveMacenski, does this need to be similar to what has been done for the theta star planner https://navigation.ros.org/configuration/packages/configuring-thetastar.html

SteveMacenski, 2022-03-01:
Yeah!

I think for some of them gifs would be better (like DWB / RPP / Waypoint) with the debug topics shown in Rviz to kind of show what's going on under the hood. 

Others like planners, an image tells the story well enough. 

It also might be nice to add a few sentences about how each of the methods works at the top of the page. I think I started to do that for the newer packages, but I think many still just have straight "here are the parameters" without any explanation on the page itself how the algorithm works, at a very brief high level (maybe with an appropriate citation?)

NeoKish, 2022-03-02:
Thanks for the additional info. Yeah I see that description part been done for waypoint follower and RPP. I will try to add the description in the remaining ones.

padhupradheep, 2022-03-25:
@SteveMacenski do you like to[ have this GIF ](https://neobotix-docs.de/ros/_images/send_multiple.gif) for the waypoint follower?

padhupradheep, 2022-03-25:
For RPP, this one might help:

[image]

SteveMacenski, 2022-03-25:
> @SteveMacenski do you like to[ have this GIF ](https://neobotix-docs.de/ros/_images/send_multiple.gif)for the waypoint follower?

I think you should make a video of setting 3 waypoints and showing the process of it doing all 3 waypoints before completing. Then speed it up to some reasonable duration and then  convert into a gif (or youtube video, that would be fine too if its being embedded into the documentation). Just showing the setting and starting to go to first isn't a full demo of the behavior, especially since navigate through poses is similar but with a subtle difference for new users.

The RPP looks like a still image?

SteveMacenski, 2022-04-29:
Got it, I see now! I think that's the right idea, but a bit more travel would be nice (e.g. have it drive around the map a bit) so that they can get a feel for the behavior in more than a couple of meters. If the video ends up being too long, we can always speed it up 2x or something so that its more reasonable in duration but still get a good overview of it. 

I like that you included the reversing though!

NeoKish, 2022-05-03:
Hi @SteveMacenski , let me know if this works for NavFn
[image]

And whether the following GiF would be okay for waypoint follower. 
Speed is at 2X. Duration is 31 secs.
[image]

SteveMacenski, 2022-05-03:
We're aiming to use maps that are more "realistic" than the turtlebot sandbox map, but that's the right spirit! 

Similar comments as from above on making sure to only have the rviz visualization part of the window (e.g. don't include the displays panel and other things) so that its front & center.  I would also turn off the particle filter to not distract from the point