ros-navigation/docs.nav2.org #277: Documentation regarding breaking change in AMCL `robot_model_type` parameter between galactic and current main

moooeeeep, 2022-02-11:
I just stumbled accross the breaking change to specify the AMCL `robot_model_type` parameter by key (up to galactic) vs by plugin class type (in current main).

The online published documentation is generated for version `latest` (current main), while most users will probably start out with the package version installed via foxy or galactic binary packages. Thus they will likely be confused when they their AMCL crashes because they provide the documented values for that parameter. (I think there should be better error handling btw.)

I couldn't find a specific galactic version of the docs. However I think with breaking changes between these versions, it might make sense to consider this. If going forwards with a single version of documentation, I think it should state the valid parameter values for earlier versions too. (Or attempt to maintain backwards compatibility, which seems possible in this case.)

SteveMacenski, 2022-02-11:
I can definitely appreciate the frustration, unfortunately we have very limited time to do things and if we spent hours and hours maintaining multiple parallel tracks of documentation we'd have that much less time to create new features, tests, or other documentation that is broadly applicable. 

> If going forwards with a single version of documentation, I think it should state the valid parameter values for earlier versions too.

 I think that's a fair compromise. If you'd be open to submitting that PR with the clarification on the configuration guide for that AMCL parameter about the old options, I'd merge it. I'm not sure how many others I can think of that we've changed so fundamentally, but I'm sure there's a couple of others

moooeeeep, 2022-02-14:
No problem. 

I prepared a draft: https://github.com/ros-planning/navigation.ros.org/pull/278

closed 2022-02-14 by SteveMacenski