ros-navigation/docs.nav2.org #306: Inquiry

YXGuan, 2022-05-01:
Hi there,

Is there a github page with source code for[ 2021 Summer of Code Semantic Integration](https://github.com/ros-planning/navigation.ros.org/blob/master/2021summerOfCode/projects/semantics.rst)?

Anything you can share would be a huge help : )

SteveMacenski, 2022-05-02:
Not much to be discussed right now -- there's not been a great deal of recent progress

closed 2022-05-02 by SteveMacenski

YXGuan, 2022-05-04:
Hi sir,

I think there can be a lot of potential in segmentation based ROS navigation, but seems like there's not much existing materials on that (except [this](https://docs.nvidia.com/isaac/isaac/packages/freespace_dnn/doc/freespace_segmentation.html#quick-start) from Nvidia).

What would be your recommendations to get started on this project? I have a little bit experience with ROS, but have no experience with semantic segmentation.

Thank you Steve,