ros-navigation/docs.nav2.org #319: Not able to build

padhupradheep, 2022-06-02:
Something is fishy, when I try to build on ubuntu 22.04

```
WARNING: Invalid configuration value found: 'language = None'. Update your configuration to a valid langauge code. Falling back to 'en' (English).

Exception occurred:
  File "/home/pradheep/.local/lib/python3.10/site-packages/breathe/node_factory.py", line 5, in <module>
    import sphinx.ext.mathbase
ModuleNotFoundError: No module named 'sphinx.ext.mathbase'
The full traceback has been saved in /tmp/sphinx-err-v_0nnovb.log, if you want to report the issue to the developers.
```

SteveMacenski, 2022-06-02:
Did you not install all necessary dependencies? It looks like `mathbase` just needs to be installed?

padhupradheep, 2022-06-07:
but it should be installed as part of the sphinx core right?

That is what the documentation says: https://www.sphinx-doc.org/en/master/usage/extensions/math.html

padhupradheep, 2022-06-09:
So, the solution would be to use breathe 4.28.0 ! I'll give a PR after testing it on 20.04 as well

closed 2022-06-09 by SteveMacenski