ros-navigation/docs.nav2.org #322: Mistake in URDF - Physical Properties on Website

austin-InDro, 2022-06-03:
```xacro
    <collision>
      <geometry>
        <box size="${base_length} ${base_width} ${base_height}"/>
      </geometry>
    </collision>

    <xacro:box_inertia m="15" w="${base_width}" d="${base_length}" h="${base_height}"/>
```
Should be 
```xacro
    <collision>
      <geometry>
        <box size="${base_width} ${base_length} ${base_height}"/>
      </geometry>
    </collision>

    <xacro:box_inertia m="15" w="${base_width}" d="${base_length}" h="${base_height}"/>
```
To follow the tutorial.

austin-InDro, 2022-06-03:
It's missing in the current master branch (which I find weird) https://github.com/ros-planning/navigation.ros.org/blob/master/setup_guides/urdf/setup_urdf.rst

So I can't submit a PR to fix unless I add all of it (which I can do if you want)

On the site, https://navigation.ros.org/setup_guides/urdf/setup_urdf.html, there is a section on adding the box inertias. However that xacro code isn't present in the git repo

austin-InDro, 2022-06-03:
Oh okay, didn't know that! I will do that then. Thanks :)

austin-InDro, 2022-06-03:
Sorry my mistake... It looks fine. I must have made the mistake on my end then looked at something else. No need to change.

closed 2022-06-03 by austin-InDro

SteveMacenski, 2022-06-03:
OK! No worries, thanks for the report either way :-)