ros-navigation/docs.nav2.org #337: First-Time Robot Setup Guide » Setting Up Sensors: costmap not publishing

makino-yyf202xz, 2022-07-25:
Found out that must set all node in one launch file in order to get costmap (instead of launch one by one as tutorial says). Am I the only one?

SteveMacenski, 2022-08-18:
What version of ROS are you using? It might be a discovery issue if you're on Foxy or older. Does this happen every time or only sometimes?

makino-yyf202xz, 2022-08-21:
I'm using humble docker image within wsl2. Also even I start them all in once, only global costmap is published. It's been a while so I don't remember clearly, but I tried turtlebot sample and humble had such similar problem but foxy went well.

SteveMacenski, 2022-08-22:
What is wsl2?

makino-yyf202xz, 2022-08-22:
WSL2 is kinda like a light virtual machine working on windows, provided by microsoft. In my situation I installed ubuntu 20.04 in window10(using this WSL2) and install docker into this ubuntu, then ran a humble image in it.

SteveMacenski, 2022-08-22:
ah ok, I haven't had a windows machine in ~6 years so I'm not up with the lingo :wink: 

My first suggestion would be to try in a non-Dockerized environment. My suspicion is that you don't have the settings right so something in the ROS-y communications are being blocked by Docker's isolation. There should be no reason why adding them all to 1 launch file vs launching N launch files would matter, beyond not launching them in the dependency order and then delaying creating the necessary nodes for so long that other's timeout looking for that information -- or network discovery issues (which could be Docker's doing).

makino-yyf202xz, 2022-08-22:
Seems a complex enviroment could be a reason that cause this kind of problem. I switched to foxy since it showed a better performence as a not best solution. Hope someday humble can as good as foxy performing different kind of enviroment. Thank you for the explaination!

closed 2022-08-23 by SteveMacenski