ros-navigation/docs.nav2.org #364: Install instructions out of date

jwallace42, 2022-11-07:
Following the directions in the readme produces the following error. 

`pip3 uninstall sphinx==3.5.0 breathe==4.28.0 sphinx_rtd_theme sphinxcontrib-plantuml jinja2==3.0.3
make html`

`Traceback (most recent call last):
  File "/home/josh/.local/bin/sphinx-build", line 5, in <module>
    from sphinx.cmd.build import main
  File "/home/josh/.local/lib/python3.10/site-packages/sphinx/cmd/build.py", line 25, in <module>
    from sphinx.application import Sphinx
  File "/home/josh/.local/lib/python3.10/site-packages/sphinx/application.py", line 32, in <module>
    from sphinx.config import Config
  File "/home/josh/.local/lib/python3.10/site-packages/sphinx/config.py", line 23, in <module>
    from sphinx.util import logging
  File "/home/josh/.local/lib/python3.10/site-packages/sphinx/util/__init__.py", line 35, in <module>
    from sphinx.util import smartypants  # noqa
  File "/home/josh/.local/lib/python3.10/site-packages/sphinx/util/smartypants.py", line 33, in <module>
    from sphinx.util.docutils import __version_info__ as docutils_version
  File "/home/josh/.local/lib/python3.10/site-packages/sphinx/util/docutils.py", line 31, in <module>
    from sphinx.util.typing import RoleFunction
  File "/home/josh/.local/lib/python3.10/site-packages/sphinx/util/typing.py", line 34, in <module>
    from types import Union as types_Union
ImportError: cannot import name 'Union' from 'types' (/usr/lib/python3.10/types.py)
`

The issue is discussed [here](https://github.com/sphinx-doc/sphinx/issues/9562). 

In order to install correctly,

SteveMacenski, 2022-11-07:
Why are you uninstalling? 

Try proposing a PR to make those changes in CI + readme for local building https://github.com/ros-planning/navigation.ros.org/blob/master/.circleci/config.yml

If it turns over correctly, then we can merge and update the pair. I just want to keep CI and local development in alignment, but yeah, its probably very outdated for the sake of keeping stability.

jwallace42, 2022-11-08:
That was a mistype :(. 

`pip3 install sphinx==4.2.0 breathe sphinx_rtd_theme sphinxcontrib-plantuml jinja2==3.0.3 make html`

I will put in a pr for this later today and hopefully the CI will build :).

AlexeyMerzlyakov, 2023-09-13:
I also have noticed `sphinx==3.5.0` dependency problems on my local Ubuntu 22.04 PC, so raising the problem. #465 attempts to fix it.

closed 2024-11-16 by SteveMacenski