ros-navigation/docs.nav2.org #423: Writing a New Controller Plugin - Out of Date?

KarolRoesner, 2023-05-19:
Hey guys, we tried making a new controller for navigation and had a problem.
Your tutorial mentions 6 functions to implement. In that case, cmake gives us the following error:

--- stderr: nav_deep_controller                             
In file included from /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp:57,
                 from /opt/ros/humble/include/class_loader/class_loader/class_loader.hpp:55,
                 from /opt/ros/humble/include/class_loader/class_loader/multi_library_class_loader.hpp:52,
                 from /opt/ros/humble/include/pluginlib/pluginlib/class_loader.hpp:38,
                 from /opt/ros/humble/include/nav2_costmap_2d/costmap_2d_ros.hpp:54,
                 from /opt/ros/humble/include/nav2_core/controller.hpp:42,
                 from /home/admin-ndietz/ros2_ws/src/nav_deep/include/nav_deep/nav_deep.hpp:23,
                 from /home/admin-ndietz/ros2_ws/src/nav_deep/src/nav_deep.cpp:1:
/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::impl::MetaObject<C, B>::create() const [with C = nav2_nav_deep_controller::NavDeepController; B = nav2_core::Controller]’:
/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:216:7:   required from here
/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:218:12: error: invalid new-expression of abstract class type ‘nav2_nav_deep_controller::NavDeepController’
  218 |     return new C;
      |            ^~~~~
In file included from /home/admin-ndietz/ros2_ws/src/nav_deep/src/nav_deep.cpp:1:
/home/admin-ndietz/ros2_ws/src/nav_deep/include/nav_deep/nav_deep.hpp:34:7: note:   because the following virtual functions are pure within ‘nav2_nav_deep_controller::NavDeepController’:
   34 | class NavDeepController : public nav2_core::Controller
      |       ^~~~~~~~~~~~~~~~~
In file included from /home/admin-ndietz/ros2_ws/src/nav_deep/include/nav_deep/nav_deep.hpp:23,
                 from /home/admin-ndietz/ros2_ws/src/nav_deep/src/nav_deep.cpp:1:
/opt/ros/humble/include/nav2_core/controller.hpp:126:16: note:     ‘virtual void nav2_core::Controller::setSpeedLimit(const double&, const bool&)’
  126 |   virtual void setSpeedLimit(const double & speed_limit, const bool & percentage) = 0;
      |                ^~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/nav_deep_controller.dir/build.make:76: CMakeFiles/nav_deep_controller.dir/src/nav_deep.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/nav_deep_controller.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< nav_deep_controller [6.42s, exited with code 2]

We fixed it by implementing the setSpeedLimit function like in the regulated_pure_pursuit_controller.
Is this function really needed or is the tutorial at https://navigation.ros.org/plugin_tutorials/docs/writing_new_nav2controller_plugin.html
deprecated?

SteveMacenski, 2023-05-19:
Yes the function is really needed so that you can use the features of nav2 to dynamically reduce speeds in certain areas or situations. It looks like when we added that feature, we did not update the tutorial likewise. @AlexeyMerzlyakov I think this is something from your work, can you handle this?

SteveMacenski, 2023-06-05:
Fixed!

closed 2023-06-05 by SteveMacenski