ros-navigation/docs.nav2.org #456: PDF file of the documentation

ARK3r, 2023-08-21:
I was wondering if it would be useful to have a pdf copy of the documentation on the website and the github repo. I know I could use it right now.

SteveMacenski, 2023-08-21:
I haven't thought about it but that could be a nice contribution if you wanted to update how the website compiles with readthedocs to also generate a PDF. If that's possible, we can figure out a way to have a link to that from the root-directory of the website for those that prefer it.

Good thought :smile:

ARK3r, 2023-08-21:
I don’t have any experience with working with read the docs but I’ve seen
it quite often downloading of different formats such as pdf and html
available in the drop up menu to select the version. I can take a look at
it tonight to see how to have it automatically generated and added to the
website.

ARK3r, 2023-08-22:
Not sure what went wrong with my last response, sorry about that, but does this look reasonable:

[navigation2.pdf](https://github.com/ros-planning/navigation.ros.org/files/12403731/navigation2.pdf)

SteveMacenski, 2024-02-02:
Instructions to be added to readme shortly in a mass update. If someone would like to have this hosted on the website, I'd be happy to merge a follow on PR that adds this process into CI and links it back to the website -- but is a `wontfix` for maintainers.

closed 2024-02-02 by SteveMacenski