ros-navigation/docs.nav2.org #467: Unable to locate package ros-rolling-navigation2 on Ubuntu 22.04

ywiyogo, 2023-09-13:
I try to use navigation2 with my installed ros2-rolling. Running `sudo apt install ros-rolling-navigation2` returns
Unable to locate package ros-rolling-navigation2
Running `sudo apt install ros-humble-navigation2` works.

Should I do extra step to install the ros-rolling-navigation2 package or it's just doesn't exist?

SteveMacenski, 2023-09-13:
there is no rolling binary. build from source.

closed 2023-09-13 by SteveMacenski

ywiyogo, 2023-09-18:
@SteveMacenski  I follow the instruction and run these commands:
```
mkdir -p ~/nav2_ws/src && cd ~/nav2_ws
git clone https://github.com/ros-planning/navigation2.git  ./src/navigation2
vcs import ./src < ./src/navigation2/tools/underlay.repos
rosdep install -y --from-paths ./src  --ignore-src
```
The `rosdep install` returns these error msgs:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_bringup: Cannot locate rosdep definition for [slam_toolbox]
```
Did I miss something?

SteveMacenski, 2023-09-18:
You should also install slam toolbox, optional if you want to use the slam launch file.

ywiyogo, 2023-09-19:
Thanks! I'm new in ros2, which slam launch file did you mean exactly? 
Isn't it maybe a dependency issue? So, the installation of nav2 needs slam_toolbox, and if I read the slam_toolbox, it needs to install nav2. In this case, I'm just trying to install the package, not to launch any launch file.
Or is the navigation2 package is part of the slam_toolbox?