ros-navigation/docs.nav2.org #468: Documentation: Bug in

smnhdw, 2023-09-19:


SteveMacenski, 2023-09-19:
> Looking into your final launch file (https://github.com/ros-planning/navigation2_tutorials/blob/master/sam_bot_description/launch/display.launch.py#L45), the GUI parameter is still existing and the launch condition of the joint_state_publisher_node is still provided.

Yes, but if its not set, then we use the robot staet publisher from https://github.com/ros-planning/navigation2_tutorials/blob/master/sam_bot_description/launch/display.launch.py#L21

>   Only remove 'joint_state_publisher_gui_node' from launch/display.launch.py

But OK! Can you open a PR to make that change? I'd love for you to get credit for your debugging!

SteveMacenski, 2024-02-02:
Mass update to docs shortly will address

closed 2024-02-02 by SteveMacenski