ros-navigation/docs.nav2.org #469: (SLAM) Navigating While Mapping: Package 'turtlebot3_bringup' not found

ywiyogo, 2023-09-20:
I tried to follow the tutorial on https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html#navigation2-with-slam.
I installed the ros-rolling-desktop, ros-rolling-turtlesim and built the navigation2 and slam_toolbox from source. 

I can run the command `ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False`  from the tutorial https://navigation.ros.org/getting_started/index.html#getting-started. 

But not the launch: `ros2 launch turtlebot3_bringup robot.launch.py`. It returns
```
Package 'turtlebot3_bringup' not found: "package 'turtlebot3_bringup' not found, searching...
```
Note,
```
ros2 launch nav2_bringup navigation_launch.py
``` 
and 
```
ros2 launch slam_toolbox online_async_launch.py
```
works on my computer.
After I search for this package. I can't find a dependency to `turtlebot3_bringup` neither in the repository of navigation2 nor in slam_toolbox.
Shall we either add the installation of this package on the tutorial page or add the dependency in one of both repository.

SteveMacenski, 2023-09-20:
> turtlebot3_bringup

We are not the turtlebot package, file it with them :smile: This is Nav2 

[image]

closed 2023-09-20 by SteveMacenski

ywiyogo, 2023-09-20:
I thought this repo is for the nav2 documentation pages., I just saw that in the requirement section of the tutorial page, there is missing turtlebot3 package.

SteveMacenski, 2023-09-20:
Ah, I clearly haven't read that tutorial recently :-) 

do `sudo apt install ros-<distro>-turtlebot3-bringup`. If the distribution doesn't have it, build from source. https://github.com/ROBOTIS-GIT/turtlebot3/tree/ros2

You can also skip all that by going to Step 4 which explains you can do the following command and launches everything you need.  The step-by-step is mostly to introduce the items required so that you know how to customize for your platform. But if you're just getting started and don't need to dissect the system, then the following gets around the TB3 bringup package completely

```
ros2 launch nav2_bringup tb3_simulation_launch.py slam:=True
```

ywiyogo, 2023-09-21:
Thanks @SteveMacenski for the hint to go to the Step 4, and also for the hint to build from the ros2 branch of the turtlebot3 repo. Currently, I'm preparing my own dockerfile for the rolling distribution. I got build errors building the turtlebot3 from the master branch, because it needs the ros2 branch instead the master.