ros-navigation/docs.nav2.org #494: Friction on Caster Link

callen5914, 2023-11-24:
I tried to implement a change to the xacro file by reducing the friction on the caster because the robot travels in the direction of the caster it jumps really bad in Gazebo... Even with 0.0 values, it still jumps pretty bad. 

Anyone else getting this behavior?
`<surface>
                <friction>
                    <ode>
                        <mu>0.0</mu>
                        <mu2>0.0</mu2>
                    </ode>
                </friction>
            </surface>
`

SteveMacenski, 2023-11-24:
That seems like something better to take up with the gazebo folks potentially

callen5914, 2023-11-25:
Fair enough... I figured if the docs proposed this URDF as a good start to use for learning the package it would function better... After all how good can nav be if your sensors can't reliably sense that environment...

closed 2023-11-25 by SteveMacenski