ros-navigation/docs.nav2.org #597: Question on the amcl algoritm, if it is working correctly

AshBastian, 2024-10-08:
Hello everybody,
I'm using ROS2 Humble with the Navigation2 Stack to move my robot around.
Everthing seem to work fine besides the Localisation from the Amcl.
It is working, to the point, where I don't get any error messages or warning from it, but the localisation is not working very well.
Here are all the parameters I use, the map and some pictures of what I mean:
```
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.0
    alpha2: 0.0
    alpha3: 0.0
    alpha4: 0.0
    alpha5: 0.0
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: true
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field_prob"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::OmniMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.005
    update_min_d: 0.025
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan
    map_topic: /map
``` 
The Map.yaml:
```
image: Map1_empty.pgm
mode: trinary
resolution: 0.05
origin: [-14.3, -14.3, 0]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.25
```
[image]

[image]
[image]

[image]
[image]

SteveMacenski, 2024-10-08:
We generally don't help with configuration assistance in the issue tracker - our docs & Robotics Stack Exchange are more meant for that. However I can give you some thoughts:

- AMCL and its parameters are almost entirely aligned entirely with Probabilistic Robotics, so its worth referencing your copy when trying to tune and understand what all the dials do. You may need to tune to optimize for a particular application and the out of the box parameter set is in the general ballpark of "ok" but by no mean optimal
- This looks like Gazebo and that you have perfect, featureless walls. While AMCL I would expect to be fine with that after it converges, note that SLAM hates this kind of thing and would struggle
- Make sure your sensor AMCL range is long enough to be able to hit the walls. I didn't count the tiles on your map for scale. It looks OK, but worth double checking. Visualize the AMCL particle cloud in rviz to cehck when it converges. It might take driving around for 10-20 seconds before it fully converges to a solution. If you just localized the robot, it might take a moment.

closed 2024-10-08 by SteveMacenski

mikeferguson, 2024-10-08:
I'm just gonna note that all 5 alpha parameters being 0 is not going to work well (or at all?) - you're basically saying that odometry is perfect and that the particle sampling shouldn't diverge at all from odometry (in which case, you might as well not even run AMCL).

SteveMacenski, 2024-10-08:
^ This is a good point, I didn't look at the config