ros-navigation/docs.nav2.org #608: Update sam_bot + first time setup guide for Modern Gazebo

SteveMacenski, 2024-11-19:


Amronos, 2024-12-01:
I am willing to work on this issue! Expect some PRs in a few days.

Amronos, 2024-12-01:
@SteveMacenski I am also going to be making a SDF file for the description, should the documentation for that be added [here](https://docs.nav2.org/setup_guides/urdf/setup_urdf.html) or should a separate tutorial be made?

SteveMacenski, 2024-12-02:
Lovely, thanks! I think a new tutorial for the SDF is appropriate so we can make sure to not confuse folks up front and have the chance to introduce the idea of an SDF separately from the URDF. I think that tutorial should be between the URDF (so they have a basic robot description working) and the Setup Odometry (so that the SDF file exists already for them to add in the modifications for the drive train / sensor simulations)! 

Thanks so much!

PR #618 (2025-01-30): Update First-Time Robot Setup Guide for Modern Gazebo
Fixes #608.

- [x] Update setup_odom.rst
- [x] Update setup_sensors.rst
- [x] Add tutorial for SDF.
closed 2025-01-30 by SteveMacenski