ros-navigation/docs.nav2.org #653: Where are Gazebo models stored?

andrejpan, 2025-03-09:
``I am following [getting_started](https://docs.nav2.org/getting_started/index.html) section, however folder directory `/opt/ros/jazzy/share/turtlebot3_gazebo/models` does not exist.

I am also getting error below, but I assume it might have an issue with gazebo models not being found...

```bash
[rviz2-7] [INFO] [1741535211.003853684] [rviz]: Message Filter dropping message: frame 'base_scan' at time 38.001 for reason 'discarding message because the queue is full'
[rviz2-7] [INFO] [1741535211.130671653] [rviz]: Message Filter dropping message: frame 'odom' at time 38.160 for reason 'discarding message because the queue is full'
[rviz2-7] [INFO] [1741535211.130839772] [rviz]: Message Filter dropping message: frame 'odom' at time 34.236 for reason 'discarding message because the queue is full'
[rviz2-7] [INFO] [1741535211.195863484] [rviz]: Message Filter dropping message: frame 'base_scan' at time 38.202 for reason 'discarding message because the queue is full'
[rviz2-7] [INFO] [1741535211.355011432] [rviz]: Message Filter dropping message: frame 'odom' at time 38.340 for reason 'discarding message because the queue is full'
[rviz2-7] [INFO] [1741535211.387442408] [rviz]: Message Filter dropping message: frame 'base_scan' at time 38.400 for reason 'discarding message because the queue is full'
[component_container_isolated-8] [INFO] [1741535211.499646889] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
```

mini-1235, 2025-03-09:
For Jazzy, I believe you do not need to set the environment variables for gazebo models directory, you can run it directly

The errors are expecting you to do a "2D pose estimate" as mentioned in this https://docs.nav2.org/getting_started/index.html#navigating section

SteveMacenski, 2025-03-09:
Please see the installation and getting started guides 🙂 it covers installing any third party dependencies for a given distribution

andrejpan, 2025-03-10:
@mini-1235 thank you for your tip, somehow things are working without setting `GAZEBO_MODEL_PATH` variable on jazzy. 

I am not a ROS expert, but there is a big discrepancy between missing "2D pose estimation" and full queue messages that the terminal says.

@SteveMacenski I literally wrote that I am following the "getting started" guide, and things are broken at step ~5 for jazzy. I am closing this ticket, hopefully other people will find "solution" here.

closed 2025-03-10 by andrejpan